HesaiTechnology/HesaiLidar_ROS_2.0 #1: timestamp

zhuYL8842, 2023-11-03:


EduJorquera, 2023-11-10:
@zhuYL8842 I'm having the same issue. DId you find any solution?

911527016, 2023-12-20:
没找到具体的解决方法所以我尝试手动修改源码，将每个点的timestamp改为当前系统时间
修改文件 HesaiLidar_ROS_2.0/src/manager/source_driver_ros2.hpp
头文件加一个  #include <rclcpp/clock.hpp>
在218行加入
  rclcpp::Clock clock(RCL_ROS_TIME);
  rclcpp::Time current_time = clock.now();
  ros_msg.header.stamp = current_time;
  int64_t nanoseconds = current_time.nanoseconds();
  double timestamp_seconds = static_cast<double>(nanoseconds) * 1e-9;
  然后在下面的for循环中插入 frame.points[i].timestamp=timestamp_seconds;
  将每个点的timstamp修改为系统时间
  再次编译运行就能得到正确的时间戳了
[image]

saurabh1002, 2024-06-12:
The solution by @911527016 solves the issue. There is just one extra change to make it compile. Can find the complete diff in #8

ARK3r, 2025-03-18:
Is there any reason this has not been merged? @changxingshuo @ForStack @zhangyu-hesaitech

kExU9853, 2025-04-08:
Exact the same issue!

changxingshuo, 2025-04-17:
设置config.yaml中的use_timestamp_type为1代表使用本地时间戳，默认为0代表使用雷达时间戳

YS-KID, 2025-04-18:
> 没找到具体的解决方法所以我尝试手动修改源码，将每个点的timestamp改为当前系统时间 修改文件 HesaiLidar_ROS_2.0/src/manager/source_driver_ros2.hpp 头文件加一个 #include <rclcpp/clock.hpp> 在218行加入 rclcpp::Clock clock(RCL_ROS_TIME); rclcpp::Time current_time = clock.now(); ros_msg.header.stamp = current_time; int64_t nanoseconds = current_time.nanoseconds(); double timestamp_seconds = static_cast(nanoseconds) * 1e-9; 然后在下面的for循环中插入 frame.points[i].timestamp=timestamp_seconds; 将每个点的timstamp修改为系统时间 再次编译运行就能得到正确的时间戳了 [image]

To set the timestamp of each point in the point cloud to the current time, we need to update the timestamps in frame.points before assigning them to point in the loop.

```
  for (size_t i = 0; i < frame.points_num; i++)
  {
    frame.points[i].timestamp=timestamp_seconds;
    LidarPointXYZIRT point = frame.points[i];
    *iter_x_ = point.x;
    *iter_y_ = point.y;
    *iter_z_ = point.z;
    *iter_intensity_ = point.intensity;
    *iter_ring_ = point.ring;
    *iter_timestamp_ = point.timestamp;
    ++iter_x_;
    ++iter_y_;
    ++iter_z_;
    ++iter_intensity_;
    ++iter_ring_;
    ++iter_timestamp_;   
  }
```

changxingshuo, 2025-04-21:
When you set use_timestamp_type to 1, it will get the local time during parsing and assign it to the timestamp in frame.points, which is ultimately convenient for you to use and meet your requirements.
For the code, please search globally for use_timestamp_type to check it out

closed 2025-05-15 by changxingshuo