HesaiTechnology/HesaiLidar_ROS_2.0 #6: Compilation errors

tseco15, 2024-05-24:
Hello, 
I am working with ROS2 Humble and Ubuntu 22.04.
Following the instructions for driver compilation, the compilation fails. There are some warnigns and finally errors.
You can find the output in the file attached.
Any suggestions?
[compilation_msgs_error.txt](https://github.com/HesaiTechnology/HesaiLidar_ROS_2.0/files/15429897/compilation_msgs_error.txt)

jemmmel, 2024-06-18:
I am getting the same error using master branch, however issue seems to be with the SDK as the same issue occurs when building the SDK using cmake
[hesai_err.txt](https://github.com/user-attachments/files/15880604/hesai_err.txt)

dholukeval, 2024-09-16:
> I am getting the same error using master branch, however issue seems to be with the SDK as the same issue occurs when building the SDK using cmake [hesai_err.txt](https://github.com/user-attachments/files/15880604/hesai_err.txt)

I am facing the same problem. I can not build the Hesai SDK nor Hesai ros driver as it is dependent on SDK. Did you find the solution?

jemmmel, 2024-09-18:
Since we are not using CUDA we simply removed it from the CMakeLists

closed 2026-10-27 by ForStack