HesaiTechnology/HesaiLidar_ROS_2.0 #11: High Latency and Low Frame Rate in RViz with XT32 LiDAR Using ROS 2 Humble SDK

thomasankara, 2024-06-18:
Hello, we have just started integrating your XT32 LiDAR into our robot. For now, we are using your SDK designed for ROS 2 Humble. My laptop is powerful, yet I observe in your test code a delay between the movement (in reality) and its visualization in RViz of about 100 to 500 ms. Additionally, I am struggling to switch to triple return mode to increase the frame rate to 20Hz; currently, I have a frame rate of 10Hz.

jessychen1016, 2024-09-10:
hi, Im encountering the same issue, have you solved it yet?

soonhong, 2025-03-12:
I'm also struggling with same issues..

ForStack, 2026-11-02:
Hi @thomasankara, @jessychen1016 — this is the same root cause as #91, which we have just answered and closed. Summarizing here since the symptoms are the same.

Point cloud conversion and publish() currently run inside the SDK decode callback. Every additional subscriber adds DDS serialization and transmission cost to that callback; once a subscriber (RViz, Foxglove or a slower consumer) falls behind, the decode thread blocks in publish(), UDP packets accumulate in the ring buffer and are eventually dropped by the kernel. That is why adding subscribers increases the lag and eventually produces packet loss. The default QoS (reliable, depth 10) also allows up to ~1 s of queued frames at 10 Hz, which on its own can explain a large part of the 100–500 ms you see.

We have split decoding from publishing: the decode callback now only snapshots the points into a bounded queue, while conversion and publishing happen on a dedicated thread. For live lidar the oldest queued frame is dropped instead of applying back-pressure to decoding. QoS is now configurable (depth/reliability/durability); for low-latency multi-subscriber setups, best_effort with depth 1 is recommended. The new version will be released at the end of 2026. Until then, the relay-node workaround described in #91 remains the safest option, and reducing the number of subscribers or the QoS depth also helps.

Two notes on the other part of your report

Measure latency at the topic level rather than in RViz; rendering is a significant and variable part of what you observe.
20 Hz is the lidar's rotation speed. Please check whether a different rotation speed has been set on the lidar's web page; this can be confirmed by capturing packets with Wireshark and checking the actual rate in the capture.
As this issue is quite old, we are closing it for now. If you still see the problem, please test with the latest version and open a new issue with the details.

closed 2026-11-02 by ForStack