HesaiTechnology/HesaiLidar_ROS_2.0 #12: If "No tf Data" shows up and you've just installed it.

Castleone98, 2024-07-05:
if there is "enable_packet_loss_tool: true" in your "config.yaml" file 
it'll executed as debug mode so rviz doesn't work

So,make "enable_packet_loss_tool: false" in your "config.yaml" file 
or Delete "enable_packet_loss_tool: true".
[image]

hasarindu-vega, 2024-07-08:
Hey,
[image]

I am currently working on connecting XT16 LiDAR sensor to ROS2 within my development environment. Despite implementing the suggested fix, I am still unable to receive any data on RViz. Below are the details of my setup:

    Operating System: Ubuntu 22.04
    ROS Version: ROS Humble

For further clarity, I have attached a screenshot illustrating the current setup and issue.

I would appreciate any guidance or suggestions you might have to resolve this matter.

hasarindu-vega, 2024-07-08:
Hey,

I am currently working on connecting XT16 LiDAR sensor to ROS2 within my development environment. Despite implementing the suggested fix, I am still unable to receive any data on RViz. Below are the details of my setup:

Operating System: Ubuntu 22.04
ROS Version: ROS Humble

For further clarity, I have attached a screenshot illustrating the current setup and issue.

I would appreciate any guidance or suggestions you might have to resolve this matter.

Thanks.

[image]

Castleone98, 2024-07-08:
It seems fine to me...Have you tried catch XT16 LIDAR's packet with wireshark? 
if XT16's packets being catched, Your computer connected with your device but some reason it doesn't work..

if not being catched,then try to check your IP addr or else.

best debuging guide that i recommend is to debug hesai SDK itself to print more info.(using print())

quner2024, 2024-08-05:
> Hey,
> 
> I am currently working on connecting XT16 LiDAR sensor to ROS2 within my development environment. Despite implementing the suggested fix, I am still unable to receive any data on RViz. Below are the details of my setup:
> 
> Operating System: Ubuntu 22.04 ROS Version: ROS Humble
> 
> For further clarity, I have attached a screenshot illustrating the current setup and issue.
> 
> I would appreciate any guidance or suggestions you might have to resolve this matter.
> 
> Thanks.
> 
> ![Screenshot from 2024-07-08 13-07-25](https://private-user-images.githubusercontent.com/158435714/346473727-e20b06a6-0172-4513-92c6-5224e955452e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI4NTA2NzYsIm5iZiI6MTcyMjg1MDM3NiwicGF0aCI6Ii8xNTg0MzU3MTQvMzQ2NDczNzI3LWUyMGIwNmE2LTAxNzItNDUxMy05MmM2LTUyMjRlOTU1NDUyZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgwNVQwOTMyNTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xODk1NDZiNjUyZjM2ODE4YzdiMWY1MGE4MTgyZjMyMWJmZGU0YTgwOGM3NzQ1OTJhNWU5MzZlZjg5ZGMyODIxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.UqXn4OwCuTC4SEppkL8fjhtz7v8zAmjm9XMITg68nG4)

Thanks to Hesai's professional help, I finally find a solution for this error.
My environment is AT128 Lidar + Ubuntu 22.04 LTS + ROS2 Humble. Also, my laptop isn't equipped with a NVIDIA Graphics Card.

What most important thing is to turn off ANY firewall like UFW(Ubuntu) or Windows Firewall(Windows). You really need to ensure that there aren't any type of block wall behind u and lidar.

Please double check! You need to completely turn off firewall first.

Then if u do not have NVIDIA Graphics Cards ,u need to go to CMakeLists.txt and find line 158 

158     **find_package(CUDA )**

And change the **WHOLE LINE** into 

158     **find_package(CUDA_FOUND )**

Finally, colcon build as official docs says:

```
  colcon build --symlink-install
  . install/local_setup.bash
```

By the way, I didn't change anything including config/config.yaml and lidar works well.

hasarindu-vega, 2024-08-29:
> > Hey,
> > I am currently working on connecting XT16 LiDAR sensor to ROS2 within my development environment. Despite implementing the suggested fix, I am still unable to receive any data on RViz. Below are the details of my setup:
> > Operating System: Ubuntu 22.04 ROS Version: ROS Humble
> > For further clarity, I have attached a screenshot illustrating the current setup and issue.
> > I would appreciate any guidance or suggestions you might have to resolve this matter.
> > Thanks.
> > ![Screenshot from 2024-07-08 13-07-25](https://private-user-images.githubusercontent.com/158435714/346473727-e20b06a6-0172-4513-92c6-5224e955452e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI4NTA2NzYsIm5iZiI6MTcyMjg1MDM3NiwicGF0aCI6Ii8xNTg0MzU3MTQvMzQ2NDczNzI3LWUyMGIwNmE2LTAxNzItNDUxMy05MmM2LTUyMjRlOTU1NDUyZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgwNVQwOTMyNTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xODk1NDZiNjUyZjM2ODE4YzdiMWY1MGE4MTgyZjMyMWJmZGU0YTgwOGM3NzQ1OTJhNWU5MzZlZjg5ZGMyODIxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.UqXn4OwCuTC4SEppkL8fjhtz7v8zAmjm9XMITg68nG4)
> 
> Thanks to Hesai's professional help, I finally find a solution for this error. My environment is AT128 Lidar + Ubuntu 22.04 LTS + ROS2 Humble. Also, my laptop isn't equipped with an NVIDIA Graphics Card.
> 
> What most important thing is to turn off ANY firewall like UFW(Ubuntu) or Windows Firewall(Windows). You really need to ensure that there aren't any type of block wall behind u and lidar.
> 
> Please double check! You need to completely turn off firewall first.
> 
> Then if u do not have NVIDIA Graphics Cards ,u need to go to CMakeLists.txt and find line 158
> 
> 158 **find_package(CUDA )**
> 
> And change the **WHOLE LINE** into
> 
> 158 **find_package(CUDA_FOUND )**
> 
> Finally, colcon build as official docs says:
> 
> ```
>   colcon build --symlink-install
>   . install/local_setup.bash
> ```
> 
> By the way, I didn't change anything including config/config.yaml and lidar works well.

It seems like the UFW is inactive in my case. I am also using a device without a GPU, so I tried changing the line 158. Below is the results.

[image]

[image]

Wireshark is capturing some packets from LiDAR, but I do not see any LiDAR pointcloud data there, and the data flow is very slow. (I assume this is some kind of a header to acknowledge LiDAR is connected?)

I also tried Hesai SDK but the result were same.

[image]

quner2024, 2024-10-26:
> > > Hey,
> > > I am currently working on connecting XT16 LiDAR sensor to ROS2 within my development environment. Despite implementing the suggested fix, I am still unable to receive any data on RViz. Below are the details of my setup:
> > > Operating System: Ubuntu 22.04 ROS Version: ROS Humble
> > > For further clarity, I have attached a screenshot illustrating the current setup and issue.
> > > I would appreciate any guidance or suggestions you might have to resolve this matter.
> > > Thanks.
> > > ![Screenshot from 2024-07-08 13-07-25](https://private-user-images.githubusercontent.com/158435714/346473727-e20b06a6-0172-4513-92c6-5224e955452e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI4NTA2NzYsIm5iZiI6MTcyMjg1MDM3NiwicGF0aCI6Ii8xNTg0MzU3MTQvMzQ2NDczNzI3LWUyMGIwNmE2LTAxNzItNDUxMy05MmM2LTUyMjRlOTU1NDUyZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgwNVQwOTMyNTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xODk1NDZiNjUyZjM2ODE4YzdiMWY1MGE4MTgyZjMyMWJmZGU0YTgwOGM3NzQ1OTJhNWU5MzZlZjg5ZGMyODIxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.UqXn4OwCuTC4SEppkL8fjhtz7v8zAmjm9XMITg68nG4)
> > 
> > 
> > Thanks to Hesai's professional help, I finally find a solution for this error. My environment is AT128 Lidar + Ubuntu 22.04 LTS + ROS2 Humble. Also, my laptop isn't equipped with an NVIDIA Graphics Card.
> > What most important thing is to turn off ANY firewall like UFW(Ubuntu) or Windows Firewall(Windows). You really need to ensure that there aren't any type of block wall behind u and lidar.
> > Please double check! You need to completely turn off firewall first.
> > Then if u do not have NVIDIA Graphics Cards ,u need to go to CMakeLists.txt and find line 158
> > 158 **find_package(CUDA )**
> > And change the **WHOLE LINE** into
> > 158 **find_package(CUDA_FOUND )**
> > Finally, colcon build as official docs says:
> > ```
> >   colcon build --symlink-install
> >   . install/local_setup.bash
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > By the way, I didn't change anything including config/config.yaml and lidar works well.
> 
> It seems like the UFW is inactive in my case. I am also using a device without a GPU, so I tried changing the line 158. Below is the results.
> 
> ![git_issue_go2lidar_1](https://private-user-images.githubusercontent.com/158435714/362569785-242c5360-3590-4f7e-922e-1ca079435d47.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk5Njg3MzQsIm5iZiI6MTcyOTk2ODQzNCwicGF0aCI6Ii8xNTg0MzU3MTQvMzYyNTY5Nzg1LTI0MmM1MzYwLTM1OTAtNGY3ZS05MjJlLTFjYTA3OTQzNWQ0Ny5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAyNlQxODQ3MTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hNmY3ZjcxZTJkZTAwN2Y0YTY4ODk3Yjg2NTNkZmZiYmE1NzhlMGQ5NDJiODc4MjQ5YTgyOGZmNGU5MGQ3YTQ5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.fEmUcGKQ0DkBM6N5-i-e0EFozmQbwjdyMrfj9k5xnyc)
> 
> ![git_issue_go2lidar_2](https://private-user-images.githubusercontent.com/158435714/362569874-22985a34-4752-4b54-b318-6dff97230b1c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk5Njg3MzQsIm5iZiI6MTcyOTk2ODQzNCwicGF0aCI6Ii8xNTg0MzU3MTQvMzYyNTY5ODc0LTIyOTg1YTM0LTQ3NTItNGI1NC1iMzE4LTZkZmY5NzIzMGIxYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAyNlQxODQ3MTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iOTc5MWEyOWRiZmU1MTIwNWY5Y2RkZGI5NWZlNGZmZTEwOWI0OTVkZGM0OGNiNWQzYmRiNTg0MWJkOTYwNTBiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.j1HxNzf8W8Di-3qJkm95v4pHKrf1oZgpjRXKNq4GRRQ)
> 
> Wireshark is capturing some packets from LiDAR, but I do not see any LiDAR pointcloud data there, and the data flow is very slow. (I assume this is some kind of a header to acknowledge LiDAR is connected?)
> 
> I also tried Hesai SDK but the result were same.
> 
> ![git_issue_go2lidar_3](https://private-user-images.githubusercontent.com/158435714/362571307-42b914bd-4f51-4067-9c77-994c8ba616fc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk5Njg3MzQsIm5iZiI6MTcyOTk2ODQzNCwicGF0aCI6Ii8xNTg0MzU3MTQvMzYyNTcxMzA3LTQyYjkxNGJkLTRmNTEtNDA2Ny05Yzc3LTk5NGM4YmE2MTZmYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAyNlQxODQ3MTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wMWM3YjBjZTFhNmU3MTIwNmZkMDY0ZjE2N2ZiNGNmMzYyMzAzYTQyYTI2Y2I4ZGQxZWVhZDI0MDY1OGMxZGQyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.2Lr9R-wMSgjstmMQ3SXccy-lmTlwP34DxXq_FgK6R2o)

Sorry for appling so late. Could u try to clone a copy which completely new and change code, then make and run? In my humble opinion, colcon build needs clean env.

wohaoshadiao, 2025-09-09:
> > > Hey,  嘿，
> > > I am currently working on connecting XT16 LiDAR sensor to ROS2 within my development environment. Despite implementing the suggested fix, I am still unable to receive any data on RViz. Below are the details of my setup:我目前正在开发环境中将 XT16 LiDAR 传感器连接到 ROS2。尽管已实施建议的修复，但我仍然无法在 RViz 上接收任何数据。以下是我的设置详情：
> > > Operating System: Ubuntu 22.04 ROS Version: ROS Humble操作系统：Ubuntu 22.04 ROS 版本：ROS Humble
> > > For further clarity, I have attached a screenshot illustrating the current setup and issue.为了进一步说明，我附上了一张屏幕截图，展示了当前的设置和问题。
> > > I would appreciate any guidance or suggestions you might have to resolve this matter.我将非常感激您为解决此事提供的任何指导或建议。
> > > Thanks.  谢谢。
> > > <img alt="Screenshot from 2024-07-08 13-07-25" width="262.1875" height="21" src="https://private-user-images.githubusercontent.com/158435714/346473727-e20b06a6-0172-4513-92c6-5224e955452e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI4NTA2NzYsIm5iZiI6MTcyMjg1MDM3NiwicGF0aCI6Ii8xNTg0MzU3MTQvMzQ2NDczNzI3LWUyMGIwNmE2LTAxNzItNDUxMy05MmM2LTUyMjRlOTU1NDUyZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgwNVQwOTMyNTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xODk1NDZiNjUyZjM2ODE4YzdiMWY1MGE4MTgyZjMyMWJmZGU0YTgwOGM3NzQ1OTJhNWU5MzZlZjg5ZGMyODIxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.UqXn4OwCuTC4SEppkL8fjhtz7v8zAmjm9XMITg68nG4">
> > 
> > 
> > Thanks to Hesai's professional help, I finally find a solution for this error. My environment is AT128 Lidar + Ubuntu 22.04 LTS + ROS2 Humble. Also, my laptop isn't equipped with an NVIDIA Graphics Card.感谢禾赛的专业帮助，我终于找到了解决这个错误的方法。我的环境是 AT128 激光雷达 + Ubuntu 22.04 LTS + ROS2 Humble。而且我的笔记本电脑没有配备 NVIDIA 显卡。
> > What most important thing is to turn off ANY firewall like UFW(Ubuntu) or Windows Firewall(Windows). You really need to ensure that there aren't any type of block wall behind u and lidar.最重要的是关闭所有防火墙，例如 UFW（Ubuntu）或 Windows 防火墙（Windows）。你真的需要确保你和激光雷达后面没有任何类型的遮挡墙。
> > Please double check! You need to completely turn off firewall first.请仔细检查！您需要先完全关闭防火墙。
> > Then if u do not have NVIDIA Graphics Cards ,u need to go to CMakeLists.txt and find line 158如果你没有 NVIDIA 显卡，你需要去 CMakeLists.txt 并找到第 158 行
> > 158 **find_package(CUDA )**158** 查找包（CUDA）**
> > And change the **WHOLE LINE** into并将**整行**改为
> > 158 **find_package(CUDA_FOUND )**158** 查找包（CUDA_FOUND）**
> > Finally, colcon build as official docs says:最后，colcon build 正如官方文档所述：
> > ```
> >   colcon build --symlink-install
> >   . install/local_setup.bash
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > By the way, I didn't change anything including config/config.yaml and lidar works well.顺便说一句，我没有改变任何东西，包括 config/config.yaml，而且激光雷达运行良好。
> 
> It seems like the UFW is inactive in my case. I am also using a device without a GPU, so I tried changing the line 158. Below is the results.在我的情况下，UFW 似乎没有启用。我使用的设备也没有 GPU，所以我尝试更改了第 158 行。结果如下。
> 
> [image]
> 
> [image]
> 
> Wireshark is capturing some packets from LiDAR, but I do not see any LiDAR pointcloud data there, and the data flow is very slow. (I assume this is some kind of a header to acknowledge LiDAR is connected?)Wireshark 正在捕获一些来自 LiDAR 的数据包，但我没有看到任何 LiDAR 点云数据，而且数据流非常慢。（我猜这是某种用于确认 LiDAR 已连接的报头？）
> 
> I also tried Hesai SDK but the result were same.我也尝试了 Hesa​​i SDK，但结果是一样的。
> 
> [image]

Hello classmate, I encountered the same problem as you. My environment is Ubuntu 22.04 ROS2 humble. I compiled successfully, modified the config file, and turned off the firewall, but the result is still not showing TF Data. If you have a solution, please reply to me. Thank you very much. Email: shuo@warnie.asia

closed 2026-10-30 by ForStack