HesaiTechnology/HesaiLidar_ROS_2.0 #15: Does not compile on ROS2 Humble

stephen-derosa, 2024-10-14:
At the moment, the `hesai_ros_driver` does not compile in ROS2 humble. Is this expected? Is there a feature branch that compiles correctly?

```
dmin@orin-dev-eddie:/workspaces/isaac_ros-dev$ colcon build --packages-select hesai_ros_driver
Starting >>> hesai_ros_driver
[Processing: hesai_ros_driver]                             
[Processing: hesai_ros_driver]                                     
[Processing: hesai_ros_driver]                                       
--- stderr: hesai_ros_driver                                          
=============================================================
-- ROS_VERSION is 2
=============================================================
=============================================================
-- ROS2 Found. ROS2 Support is turned On.
=============================================================
CMake Deprecation Warning at src/driver/HesaiLidar_SDK_2.0/libhesai/CMakeLists.txt:5 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

=============================================================
-- CUDA Found. CUDA Support is turned On.
=============================================================
CMake Deprecation Warning at /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_target_interfaces.cmake:32 (message):
  Use rosidl_get_typesupport_target() and target_link_libraries() instead of
  rosidl_target_interfaces()
Call Stack (most recent call first):
  CMakeLists.txt:262 (rosidl_target_interfaces)

cc1: warning: command-line option ‘-std=c++17’ is valid for C++/ObjC++ but not for C
cc1: warning: command-line option ‘-std=c++17’ is valid for C++/ObjC++ but not for C
cc1: warning: command-line option ‘-std=c++17’ is valid for C++/ObjC++ but not for C
cc1: warning: command-line option ‘-std=c++17’ is valid for C++/ObjC++ but not for C
cc1: warning: command-line option ‘-std=c++17’ is valid for C++/ObjC++ but not for C
cc1: warning: command-line option ‘-std=c++17’ is valid for C++/ObjC++ but not for C
cc1: warning: command-line option ‘-std=c++17’ is valid for C++/ObjC++ but not for C
cc1: warning: command-line option ‘-std=c++17’ is valid for C++/ObjC++ but not for C
cc1: warning: command-line option ‘-std=c++17’ is valid for C++/ObjC++ but not for C
cc1: warning: command-line option ‘-std=c++17’ is valid for C++/ObjC++ but not for C
cc1: warning: command-line option ‘-std=c++17’ is valid for C++/ObjC++ but not for C
cc1: warning: command-line option ‘-std=c++17’ is valid for C++/ObjC++ but not for C
cc1: warning: command-line option ‘-std=c++17’ is valid for C++/ObjC++ but not for C
cc1: warning: command-line option ‘-std=c++17’ is valid for C++/ObjC++ but not for C
cc1: warning: command-line option ‘-std=c++17’ is valid for C++/ObjC++ but not for C
cc1: warning: command-line option ‘-std=c++17’ is valid for C++/ObjC++ but not for C
cc1: warning: command-line option ‘-std=c++17’ is valid for C++/ObjC++ but not for C
cc1: warning: command-line option ‘-std=c++17’ is valid for C++/ObjC++ but not for C
/opt/ros/humble/include/rclcpp/rclcpp/logger.hpp(106): warning #970-D: the qualifier on this friend declaration is ignored
    friend Logger rclcpp::get_logger(const std::string & name);
                  ^

Remark: The warnings can be suppressed with "-diag-suppress <warning-number>"

/opt/ros/humble/include/rclcpp/rclcpp/logger.hpp(106): warning #970-D: the qualifier on this friend declaration is ignored
    friend Logger rclcpp::get_logger(const std::string & name);
                  ^

Remark: The warnings can be suppressed with "-diag-suppress <warning-number>"

/opt/ros/humble/include/rclcpp/rclcpp/network_flow_endpoint.hpp(92): warning #970-D: the qualifier on this friend declaration is ignored
    friend bool rclcpp::operator==(
                ^

/opt/ros/humble/include/rclcpp/rclcpp/network_flow_endpoint.hpp(95): warning #970-D: the qualifier on this friend declaration is ignored
    friend bool rclcpp::operator!=(
                ^

/opt/ros/humble/include/rclcpp/rclcpp/network_flow_endpoint.hpp(100): warning #970-D: the qualifier on this friend declaration is ignored
[... 114 lines cut ...]
  391 |       set_deprecated(static_cast<typename scbth::callback_type>(callback));
      |       ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                  
/opt/ros/humble/include/rclcpp/rclcpp/any_subscription_callback.hpp:408:1: note: declared here
  408 |   set_deprecated(std::function<void(std::shared_ptr<SetT>)> callback)
      | ^ ~~~~~~~~~~~~
/opt/ros/humble/include/rclcpp/rclcpp/any_subscription_callback.hpp: In instantiation of ‘rclcpp::AnySubscriptionCallback<MessageT, AllocatorT> rclcpp::AnySubscriptionCallback<MessageT, AllocatorT>::set(CallbackT) [with CallbackT = std::_Bind<void (SourceDriver::*(SourceDriver*, std::_Placeholder<1>))(std::shared_ptr<std_msgs::msg::UInt8MultiArray_<std::allocator<void> > >)>; MessageT = std_msgs::msg::UInt8MultiArray_<std::allocator<void> >; AllocatorT = std::allocator<void>]’:
/opt/ros/humble/include/rclcpp/rclcpp/subscription_factory.hpp:94:30:   required from ‘rclcpp::SubscriptionFactory rclcpp::create_subscription_factory(CallbackT&&, const rclcpp::SubscriptionOptionsWithAllocator<AllocatorT>&, typename MessageMemoryStrategyT::SharedPtr, std::shared_ptr<rclcpp::topic_statistics::SubscriptionTopicStatistics<ROSMessageType> >) [with MessageT = std_msgs::msg::UInt8MultiArray_<std::allocator<void> >; CallbackT = std::_Bind<void (SourceDriver::*(SourceDriver*, std::_Placeholder<1>))(std::shared_ptr<std_msgs::msg::UInt8MultiArray_<std::allocator<void> > >)>; AllocatorT = std::allocator<void>; SubscriptionT = rclcpp::Subscription<std_msgs::msg::UInt8MultiArray_<std::allocator<void> > >; MessageMemoryStrategyT = rclcpp::message_memory_strategy::MessageMemoryStrategy<std_msgs::msg::UInt8MultiArray_<std::allocator<void> >, std::allocator<void> >; ROSMessageType = std_msgs::msg::UInt8MultiArray_<std::allocator<void> >; typename MessageMemoryStrategyT::SharedPtr = std::shared_ptr<rclcpp::message_memory_strategy::MessageMemoryStrategy<std_msgs::msg::UInt8MultiArray_<std::allocator<void> >, std::allocator<void> > >]’
/opt/ros/humble/include/rclcpp/rclcpp/create_subscription.hpp:122:63:   required from ‘std::shared_ptr<ROSMessageT> rclcpp::detail::create_subscription(NodeParametersT&, NodeTopicsT&, const string&, const rclcpp::QoS&, CallbackT&&, const rclcpp::SubscriptionOptionsWithAllocator<AllocatorT>&, typename MessageMemoryStrategyT::SharedPtr) [with MessageT = std_msgs::msg::UInt8MultiArray_<std::allocator<void> >; CallbackT = std::_Bind<void (SourceDriver::*(SourceDriver*, std::_Placeholder<1>))(std::shared_ptr<std_msgs::msg::UInt8MultiArray_<std::allocator<void> > >)>; AllocatorT = std::allocator<void>; SubscriptionT = rclcpp::Subscription<std_msgs::msg::UInt8MultiArray_<std::allocator<void> > >; MessageMemoryStrategyT = rclcpp::message_memory_strategy::MessageMemoryStrategy<std_msgs::msg::UInt8MultiArray_<std::allocator<void> >, std::allocator<void> >; NodeParametersT = rclcpp::Node; NodeTopicsT = rclcpp::Node; ROSMessageType = std_msgs::msg::UInt8MultiArray_<std::allocator<void> >; std::string = std::__cxx11::basic_string<char>; typename MessageMemoryStrategyT::SharedPtr = std::shared_ptr<rclcpp::message_memory_strategy::MessageMemoryStrategy<std_msgs::msg::UInt8MultiArray_<std::allocator<void> >, std::allocator<void> > >]’
/opt/ros/humble/include/rclcpp/rclcpp/create_subscription.hpp:190:109:   required from ‘std::shared_ptr<ROSMessageT> rclcpp::create_subscription(NodeT&, const string&, const rclcpp::QoS&, CallbackT&&, const rclcpp::SubscriptionOptionsWithAllocator<AllocatorT>&, typename MessageMemoryStrategyT::SharedPtr) [with MessageT = std_msgs::msg::UInt8MultiArray_<std::allocator<void> >; CallbackT = std::_Bind<void (SourceDriver::*(SourceDriver*, std::_Placeholder<1>))(std::shared_ptr<std_msgs::msg::UInt8MultiArray_<std::allocator<void> > >)>; AllocatorT = std::allocator<void>; SubscriptionT = rclcpp::Subscription<std_msgs::msg::UInt8MultiArray_<std::allocator<void> > >; MessageMemoryStrategyT = rclcpp::message_memory_strategy::MessageMemoryStrategy<std_msgs::msg::UInt8MultiArray_<std::allocator<void> >, std::allocator<void> >; NodeT = rclcpp::Node; std::string = std::__cxx11::basic_string<c [... cut ...]
      |       ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                  
/opt/ros/humble/include/rclcpp/rclcpp/any_subscription_callback.hpp:408:1: note: declared here
  408 |   set_deprecated(std::function<void(std::shared_ptr<SetT>)> callback)
      | ^ ~~~~~~~~~~~~
---
Finished <<< hesai_ros_driver [1min 36s]

Summary: 1 package finished [1min 37s]
  1 package had stderr output: hesai_ros_driver
admin@orin-dev-eddie:/workspaces/isaac_ros-dev$ 

```

quner2024, 2024-10-26:
> At the moment, the `hesai_ros_driver` does not compile in ROS2 humble. Is this expected? Is there a feature branch that compiles correctly?
>
> ```
> dmin@orin-dev-eddie:/workspaces/isaac_ros-dev$ colcon [...]

I'm also working on ROS2 Humble and Hesai's code works well for me. However, I didn't use CUDA. So I suggest try to see issues before and disable cuda first. If that works, the problem may be easier to look for.

stephen-derosa, 2024-10-28:
my initial post here was wrong -- the code compiles, but with a ton of warnings.

changxingshuo, 2025-04-18:
Yes, compatibility issues can cause some warnings, we'll try to fix that!

closed 2025-04-18 by changxingshuo