HesaiTechnology/HesaiLidar_ROS_2.0 #19: How to get data whose echo mode is dual echo

Natene, 2024-12-19:
How to get data whose echo mode is dual echo

Natene, 2024-12-19:
How to set the scanning frame rate to 20 HZ

changxingshuo, 2025-04-18:
1. In the ptc_tool.cc file in the SDK, the SetReturnMode function can modify the return mode of the radar, but it is necessary to confirm whether the radar supports the modification and which modes are supported according to the radar specification, and therefore to confirm the input parameter return_mode.
2. Please contact sales and ask if the radar you are using supports modification to 20HZ and how to modify it.
You can refer to the ptc_tool.cc file, SetSpinSpeed function can modify the RPM as a way to improve the frame rate. Note: Lidar may not support

closed 2025-07-01 by changxingshuo