HesaiTechnology/HesaiLidar_ROS_2.0 #20: Node crashes when PCAP file is too large

EyedBread, 2025-02-07:
Memory becomes overflown which causes the system to kill the process. Maybe a fix that doesn't load in the entire PCAP at once would help.

changxingshuo, 2025-04-18:
Currently loading too big pcap will lead to lack of memory and finally crash the program, please use script to cut too big pcap file and then use it again.

closed 2025-04-18 by changxingshuo

EyedBread, 2025-05-09:
Ok, but stitching up big PCAP files, running the Hesai ROS package through each of them, then stitching the rosbags back together is extra unnecessary redundancy and a hassle. There are numerous techniques for reading in a large pcap file piece by piece into the software without the additional manual overhead.

changxingshuo, 2025-05-12:
Okay, we will follow up to optimize this section, thanks for your suggestion!