HesaiTechnology/HesaiLidar_ROS_2.0 #33: start time and end time set to 0.0000 when changing fov_start and fov_end

marco-monforte, 2025-04-30:
When I change the fov_start and fov_end fields in the config file for my XT16, the start time and end time logged when functioning are set to 0.000000 instead of something like 1746029829.769249.

This is my config file:

```
lidar:
  - driver:
      pcap_play_synchronization: true  # pcap play rate synchronize with the host time
      udp_port: 2368                   #UDP port of lidar
      ptc_port: 9347                   #PTC port of lidar
      device_ip_address: 192.168.123.20 # [MODIFY HERE] host_ip_address
      group_address: 255.255.255.255
      pcap_path: "Your pcap file"      #The path of pcap file
      correction_file_path: "Your correction file path" #The path of correction file
      firetimes_path: "Your firetime file path"         #The path of firetimes file
      source_type: 1                                 #The type of data source, 1: real-time lidar connection, 2: pcap, 3: packet rosbag, 4: serial    
      frame_start_azimuth: 0         #Frame azimuth for Pandar128, range from 1 to 359, set it less than 0 if you do not want to use it
      standby_mode: -1                 #The standby mode: [-1] is invalit [0] in operation [1] standby
      speed: -1                        #The speed: [-1] invalit, you must make sure your set has been supported by the lidar you are using
      #lidar configuration using serial communication
      rs485_com: "Your serial port name for receiving point cloud"  # if using JT16, Port to receive the point cloud
      rs232_com: "Your serial port name for sending cmd"            # if using JT16, Port to send cmd
      correction_save_path: ""         # if using JT16, turn on when you need to store angle calibration files(from lidar)
      #transform param
      x: 0
      y: 0
      z: 0
      roll: 0
      pitch: 0
      yaw: 1.57079   # [MODIFY HERE] 
      #decoder param
      use_timestamp_type: 1           # [MODIFY HERE] 0 use point cloud timestamp; 1 use receive timestamp
      #fov config, [fov_start, fov_end] range [1, 359], [-1, -1] means use default
      fov_start: -1   #  [MODIFY HERE] 
      fov_end:  -1    #  [MODIFY HERE] 
      enable_packet_loss_tool: true     # enable the udp packet loss detection tool
      distance_correction_lidar_flag: false # set to true when optical centre correction needs to be turned on
      xt_spot_correction: false             # Set to TRUE when XT S point cloud layering correction is required
      device_udp_src_port: 0                # Filter point clouds for specified source ports in case of multiple lidar, setting >=1024
      device_fault_port: 0                  # Filter fault message for specified source ports in case of multiple lidar, setting >=1024
    ros:
      ros_frame_id: hesai_xt16                     # [MODIFY HERE] Frame id of packet message and point cloud message
      # ros_recv_correction_topic: /lidar_corrections #Topic used to receive corrections file from rosbag
      ros_recv_packet_topic: /lidar_packets         #Topic used to receive lidar packets from rosbag
      ros_send_packet_topic: /lidar_packets         #Topic used to send lidar raw packets through ROS
      ros_send_point_cloud_topic: /lidar_points     #Topic used to send point cloud through ROS
      # ros_send_correction_topic: /lidar_corrections #Topic used to send correction through ROS
      # ros_send_firetime_topic: /lidar_firetime
      ros_send_packet_loss_topic: /lidar_packets_loss
      ros_send_imu_topic: /lidar_imu                #Topic used to send lidar imu message
      # Topic used to monitor packets loss condition through ROS
      # ros_send_ptp_topic: /lidar_ptp
      # Topic used to send PTP lock status, offset through ROS
      send_packet_ros: true                         #true: Send packets through ROS 
      send_point_cloud_ros: true                    #true: Send point cloud through ROS    
```

Do you know what I can do?

changxingshuo, 2025-05-07:
The fov_start and fov_end fields assign a portion of the point output to 0, including the timestamp.
The current start and end times use the time of the first and last points, which may be outside of fov and therefore assigned to 0.
The simple solution is to stop using the timestamps of the first and last points. Modify it to a traversal that finds the timestamp of the first non-zero point.
Subsequent versions will fix this. The way to do this is to add two timestamp variable records directly in the frame, which are recorded at parse time. Avoid such a problem.

Translated with DeepL.com (free version)

changxingshuo, 2025-05-12:
@marco-monforte 
XT parsing belongs to 6_1, go to the udp6_1_parser.cc file and find 
`if (fov_transfer < frame.config.fov_start || fov_transfer > frame.config.fov_end) `
This line of code, where the memset function assigns the indicated value of the point to 0, please add a line of code after the memset function, before continue, 
`setTimestamp(frame.points[point_index], double(frame.sensor_ timestamp[packet_index]) / kMicrosecondToSecond);`
the time information will be updated and retained

marco-monforte, 2025-05-12:
It worked, thanks a lot!