HesaiTechnology/HesaiLidar_ROS_2.0 #36: No imu values with XT16

marco-monforte, 2025-05-15:
Hi,

I'm trying to get the IMU values from the topic /lidar_imu, but they seem to not be published. Where can I double-check or enable it in the code?

This is my config file:
```
lidar:
  - driver:
      pcap_play_synchronization: true  # pcap play rate synchronize with the host time
      udp_port: 2368                   #UDP port of lidar
      ptc_port: 9347                   #PTC port of lidar
      device_ip_address: 192.168.123.20 # [MODIFY HERE] host_ip_address
      group_address: 255.255.255.255
      pcap_path: "Your pcap file"      #The path of pcap file
      correction_file_path: "Your correction file path" #The path of correction file
      firetimes_path: "Your firetime file path"         #The path of firetimes file
      source_type: 1                                 #The type of data source, 1: real-time lidar connection, 2: pcap, 3: packet rosbag, 4: serial    
      frame_start_azimuth: 0         #Frame azimuth for Pandar128, range from 1 to 359, set it less than 0 if you do not want to use it
      standby_mode: -1                 #The standby mode: [-1] is invalit [0] in operation [1] standby
      speed: -1                        #The speed: [-1] invalit, you must make sure your set has been supported by the lidar you are using
      #lidar configuration using serial communication
      rs485_com: "Your serial port name for receiving point cloud"  # if using JT16, Port to receive the point cloud
      rs232_com: "Your serial port name for sending cmd"            # if using JT16, Port to send cmd
      correction_save_path: ""         # if using JT16, turn on when you need to store angle calibration files(from lidar)
      #transform param
      x: 0
      y: 0
      z: 0
      roll: 0
      pitch: 0
      yaw: 1.57079   # [MODIFY HERE] 
      #decoder param
      use_timestamp_type: 1           # [MODIFY HERE] 0 use point cloud timestamp; 1 use receive timestamp
      #fov config, [fov_start, fov_end] range [1, 359], [-1, -1] means use default
      fov_start: 45   #  [MODIFY HERE] 
      fov_end:  315    #  [MODIFY HERE] 
      enable_packet_loss_tool: true     # enable the udp packet loss detection tool
      distance_correction_lidar_flag: false # set to true when optical centre correction needs to be turned on
      xt_spot_correction: false             # Set to TRUE when XT S point cloud layering correction is required
      device_udp_src_port: 0                # Filter point clouds for specified source ports in case of multiple lidar, setting >=1024
      device_fault_port: 0                  # Filter fault message for specified source ports in case of multiple lidar, setting >=1024
    ros:
      ros_frame_id: hesai_xt16                     # [MODIFY HERE] Frame id of packet message and point cloud message
      # ros_recv_correction_topic: /lidar_corrections #Topic used to receive corrections file from rosbag
      ros_recv_packet_topic: /lidar_packets         #Topic used to receive lidar packets from rosbag
      ros_send_packet_topic: /lidar_packets         #Topic used to send lidar raw packets through ROS
      ros_send_point_cloud_topic: /lidar_points     #Topic used to send point cloud through ROS
      # ros_send_correction_topic: /lidar_corrections #Topic used to send correction through ROS
      # ros_send_firetime_topic: /lidar_firetime
      ros_send_packet_loss_topic: /lidar_packets_loss
      ros_send_imu_topic: /lidar_imu                #Topic used to send lidar imu message
      # Topic used to monitor packets loss condition through ROS
      # ros_send_ptp_topic: /lidar_ptp
      # Topic used to send PTP lock status, offset through ROS
      send_packet_ros: true                         #true: Send packets through ROS 
      send_point_cloud_ros: true                    #true: Send point cloud through ROS    
```

Thanks

MatteGombia, 2025-05-17:
There was a bug where in cuda the code wasn't checking the right variable and the callback was never called. And without Cuda the imu was published many times with the same values. I made a pull request (https://github.com/HesaiTechnology/HesaiLidar_ROS_2.0/issues/24) in the sdk if you want to check it out, you can try that version until they accept the pull request. Let me know if it solves the problem

changxingshuo, 2025-05-20:
There is no IMU data for the XT Lidar. @marco-monforte

marco-monforte, 2025-05-20:
Oooooh, that makes more sense! Thanks @changxingshuo !

closed 2025-05-20 by marco-monforte