HesaiTechnology/HesaiLidar_ROS_2.0 #57: Submodule SDK head not found

AAArthurLiu, 2025-07-07:
Default submodule HEAD is missing.
"""
fatal: remote error: upload-pack: not our ref 564a57ad74e9b1827fccbcc6d9fa3f0052686b9d
fatal: Fetched in submodule path 'src/driver/HesaiLidar_SDK_2.0', but it did not contain 564a57ad74e9b1827fccbcc6d9fa3f0052686b9d. Direct fetching of that commit failed.
"""

edonisraci94, 2025-07-08:
I guess the ref 564... is not available anymore. I fixed it by rm -rf src/driver/HesaiLidar_SDK_2.0 afterwards you can git submodule update --init --recursive 
I guess a fix will come soon :)

AAArthurLiu, 2025-07-08:
> I guess the ref 564... is not available anymore. I fixed it by rm -rf src/driver/HesaiLidar_SDK_2.0 afterwards you can git submodule update --init --recursive I guess a fix will come soon :)

Yea, I did the same. A simple official fix should do.

AAArthurLiu, 2025-07-10:
> Sorry for that strange line in the command text btw 😅

Could refer to the official fix: https://github.com/HesaiTechnology/HesaiLidar_ROS_2.0/pull/59

Diogofcpereira14, 2025-07-10:
Btw does PR:#60 fix this?

closed 2025-07-25 by changxingshuo